rossengeorgiev/aprs-python #19: I dont want to connect to aprs.is

mathisono, 2016-08-15:
I dont want to connect to aprs.is. I working on a action script that message chats with aprs stations. I dont quite understand how to redirect the code to connect to xastir, or a local aprs server.

rossengeorgiev, 2016-08-15:
Hi @mathisono, I am not familiar with xastir. Currently the module supports connecting only to core servers, which run `javAPRSSrvr` or `aprsc`. The `IS` object has a `host` parameter to specify the hostname/address. The default is `rotate.aprs.net`.

mathisono, 2016-08-16:
Well that toobad... Im running this on a mesh network, so I dont expect to have a path to the internet but the Aprs server might. Am I correct that I can define arguments  in this function (self, callsign, passwd="-1", host="rotate.aprs.net", port=10152) ???? or hard set it in /usr/local/lib/python2.7/dist-packages/aprslib/inet.py ?   I will setup a javAPRSsrvr and see if i can get this code to connect.

Xastir connects to aprs.is so i dont know, but would suspect they do talk the same way but there is something up with the code preventing the user from defining a different server other than aprs.is.  
This is how xastir gets connected to https://xastir.org/index.php/Server_Port   

Thoughts?

rossengeorgiev, 2016-08-16:
You don't need to edit the library. Here is an example

```python
import aprslib

def receive_packet(line):
    print line

client = aprslib.IS("AB1CDE", host="127.0.0.1", port=2023)
client.connect(blocking=True)
client.consumer(receive_packet, raw=True)

Reading your link suggest that it might work using xastir's port 2023. Try it out.

mathisono, 2016-08-16:
It connects but aprslib.inet thinks it has failed....

INFO:aprslib.inet:Attempting connection to localhost:2023
INFO:aprslib.inet:Connected to ('127.0.0.1', 2023)
DEBUG:aprslib.inet:Banner: # Welcome to Xastir's server port, callsign: KJ6DZB-4
INFO:aprslib.inet:Sending login information
DEBUG:aprslib.inet:Server: K6PHD-9>APT311,WB6TMS-6*,WIDE1*,WIDE2-1:!3805.24N/12233.18Wk321/016/A=-00000/HooDatDonDar
ERROR:aprslib.inet:Failed to login
INFO:aprslib.inet:Retrying connection is 30 seconds.

then it just repeats every 30 sec

rossengeorgiev, 2016-08-16:
@mathisono, I've released a new version, update your library and use the `skip_login` parameter.

```python
client = aprslib.IS("AB1CDE", host="127.0.0.1", port=2023, skip_login=True)

mathisono, 2016-08-17:
that did the trick! It might help the next person that wonders along to update the readme on how to do this.
 Next to filter packets based on a list, extract messages, add them via an API. Then via-versa back to the air.  
Thanks!

closed 2016-10-28 by rossengeorgiev