rossengeorgiev/aprs-python #21: Generating APRS messages?

mfhepp, 2016-10-27:
Hi, 
I am looking for a library for *generating* APRS messages in Python and I saw that at this projects supports this for at least a few basic message types.

Do you have plans to add support for encoding data into APRS payload syntax? Or do you know other Python libraries that do?

Thanks!

rossengeorgiev, 2016-10-28:
Currently, you can only generate a position report via `aprslib.packets.PositionReport`. I'm not sure what you mean by "APRS payload syntax" and I'm not aware of other python libraries that can do what you are looking for.

mfhepp, 2016-10-28:
Thanks! With "APRS payload syntax" I meant all the other forms of APRS messages or information field content, like telemetry.

closed 2016-10-28 by mfhepp

rossengeorgiev, 2016-10-28:
Sure, at some point those will be available