rossengeorgiev/aprs-python #22: Connectin with inet.py

bgmarble, 2017-01-10:
Using python3 and aprslib 6.45
I am unsuccessful at connecting to an APRS2 server except when I use literal strings in the function call.
I have verified that each part of my tuple is a string before I pass it to .IS.
As you can see it tries to login to the default APRS2 location not to args in the tuple.
I have included a commented line of code that works that includes the string literals.
Your suggestions on what I can change to make the .IS function call work in this situation.
Thanks,
Brian

Here is the code
login = (settings.APRS_USER, settings.APRS_PASSCODE, settings.APRS_SERVER_HOST,     settings.APRS_SERVER_PORT, False)
print (settings.APRS_USER)
print (type(settings.APRS_USER))
print ("login - " , login)
logging.basicConfig(level=logging.DEBUG) # level=10	
AIS = aprslib.IS(login)
#AIS = aprslib.IS('KG7AFQ-1', '18869', 'socal.aprs2.net', 14580, False) #this works
AIS.connect(blocking=False)

Here is the output
KG7AFQ-1
<class 'str'>
login -  ('KG7AFQ-1', '18869', 'socal.aprs2.net', '14580', False)
INFO:aprslib.inet.IS:Attempting connection to rotate.aprs.net:10152
INFO:aprslib.inet.IS:Connected to ('212.11.82.7', 10152)
DEBUG:aprslib.inet.IS:Banner: # aprsc 2.1.4-g408ed49
INFO:aprslib.inet.IS:Sending login information
DEBUG:aprslib.inet.IS:Server: # Invalid username format
ERROR:aprslib.inet.IS:Failed to login
exception in clsAPRSConnection
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/aprslib/inet.py", line 289, in _send_login
    _, _, callsign, status, _ = test.split(' ', 4)
ValueError: not enough values to unpack (expected 5, got 4)

rossengeorgiev, 2017-01-10:
You are passing the login `tuple` as a variable.

You can unpack using `*`, like this `aprslib.IS(*login)`. That is equivalent to:  
`aprslib.IS(settings.APRS_USER, settings.APRS_PASSCODE, settings.APRS_SERVER_HOST, settings.APRS_SERVER_PORT, False)`

bgmarble, 2017-01-10:
Rossen,
 
Wow, 
 
It worked like a charm.  First time.
 
Thank you,
Brian

closed 2017-01-11 by rossengeorgiev