rossengeorgiev/aprs-python #23: Parsing Message Acks

bgmarble, 2017-01-28:
Using python3 and aprslib 6.45

Moving my server app to Aprs-Python from libfap-python to make it Python3 compatible. I deal exclusively with APRS messages. 

The challenge I have is when an 'ack' message is rec'd from a client.  It is parsing it as a 'message' with 'message_text'. I believe it should be a 'message' with 'response'. I have looked through old commits and issues with no indications of related issues. 

Here is a parsed packet showing my challenge;
 {'via': 'T2SOCAL', 'format': 'message', 'to': 'APWW10', 'raw': 'KG7AFQ-11>APWW10,TCPIP*,qAC,T2SOCAL::BAKERSMS :ack38648', 'from': 'KG7AFQ-11', 'message_text': 'ack38648', 'path': ['TCPIP*', 'qAC', 'T2SOCAL'], 'addresse': 'BAKERSMS'}

I use APRSDroid, YAAC and APRSIS32 to test my application.

Thanks for your time

rossengeorgiev, 2017-01-28:
Hi,
I have no idea how libfap-python works. This package has not attempted to mimic the FAP library, so functionality can be different and not available. It simply parses APRS into a pythonic data format and has no notion of whether the message was send or received.

bgmarble, 2017-01-28:
My apologies for not explaining my challenge better. It only involves APRS-Python.
 
This is a parsed aprslib packet.
{'via': 'T2SOCAL', 'format': 'message', 'to': 'APWW10', 'raw': 'KG7AFQ-11>APWW10,TCPIP*,qAC,T2SOCAL::BAKERSMS :ack38648', 'from': 'KG7AFQ-11', 'message_text': 'ack38648', 'path': ['TCPIP*', 'qAC', 'T2SOCAL'], 'addresse': 'BAKERSMS'}
 
Based on the raw element, should it not show a response and msgno and not show message_text because it is an ack message.
 
Thanks,
Brian

rossengeorgiev, 2017-01-28:
Ok, I got what you mean. I've fixed the code, and released a new version on pypi

closed 2017-01-28 by rossengeorgiev