rossengeorgiev/aprs-python #25: Is aprslib Intended to Generate Valid APRS packets?

kb1lqc, 2017-03-05:
Per the [APRS Protocol](http://www.aprs.org/doc/APRS101.PDF):

[image]

[image]

However with the following latitude and longitude the resulting string when using `aprslib.packets.PositionReport()` is:

```
3359.7036
11826.9392
2017-03-05 01:13:28,890 - APRS - INFO - KB1LQC-0>KB1LQC-1:!335942.22N/1182656.35En/A=000023Faraday Wireless Node
```

However, NMEA GGA data is sent down as ddMM.mmmm and dddMM.mmmm:
[image]

This means that we shouldn't have to do any conversions, just rounding and some zero stuffing to get

latitude = ddMM.mm
Longitude = dddMM.mm

A correct string sent to APRS-IS with non aprslib code with similar coordinates is:
`KB1LQC-2>GPSFDY,QAR,KB1LQC-1,qAS,KB1LQC:!3359.70N/11826.94Wn.../000/A=000037Faraday Wireless Node`

This shows up exactly where I'd expect it to. I cannot get aprslib to generate APRS compliant strings. Is this the intended function?

rossengeorgiev, 2017-03-05:
The values you input should be valid decimal degrees, not degree min sec

kb1lqc, 2017-03-05:
@rossengeorgiev yes per the code that is the case. I am curious if you are interested in supporting or ever intended to support NMEA data from a GPS which would necessitate a different algorithm.

rossengeorgiev, 2017-03-05:
> @rossengeorgiev yes per the code that is the case. 

Nope. You are providing ddmm.mmmm. The acceptable range is -90 to 90, and -180 to 180.
The code is very simple and it should be straight forward to implement property that get/sets coordinates in ddmm.mmmm format

> support NMEA data from a GPS

Generating NMEA? Seems out of scope. Parsing NMEA, maybe.

kb1lqc, 2017-03-06:
I never meant generating NMEA I meant parsing it. It would seem to me that if you want to generate strings to send into the APRS-IS system or over actual radio links you would want to convert from NMEA formatted GPS strings which is ddmm.mmmm and dddmm.mmmm for lat and long respectively.

I wrote a [quick NMEA to decimal degrees converter](https://github.com/kb1lqc/Faraday-Software/blob/aprslib/Applications/APRS/aprs.py#L165) in my APRS application. **I'd also be happy to push this or something like it back into aprslib**. I've confirmed that the resulting decimal degrees matches me input NMEA data.

## Sending aprslib Packets to APRS-IS
I have generated strings with aprslib to send to APRS-IS but no dice. Something is wrong when I upload to APRS-IS using aprslib strings as my original code works and uploads fine.

```
KB1LQC-0>KB1LQC-0:!3403.13N/11814.62Wn/A=000033Faraday Wireless Node
```

This doesn't seem to work. It's supposed to put a packet onto APRS-IS which essentially means that the remote station and local station are the same (this is the 'access point'). My coordinates seem OK and so does comments/altitude. Any idea why it might not be accepted? Have you successfully sent aprslib generated packets into APRS-IS?

rossengeorgiev, 2017-03-07:
There is already code for converting ddmm.mm to decimal degrees. It can be refactored into a separate function and reused where needed.

> Sending aprslib Packets to APRS-IS

Hard to tell without looking at code. Are you maybe connected to a receive only port?

kb1lqc, 2017-03-07:
@rossengeorgiev where is the NMEA to ddmm.mm code in `aprslib`?

kb1lqc, 2017-03-07:
OK so the problem was that I was using my own sendall connection to APRS-IS which works with my old non-aprslib code. The thought was that the socket was the same and I just used your library to generate the APRS string. This did not work.

Using your socket connection to APRS-IS with your own generated string allowed the program to actually work. I much not have seen how this is any different using str() on your position packet to simply get the string. Now that I'm using your connection code it's working fine... weird.

So I will wait to hear back about NMEA to ddmm.mm conversion in aprslib and then close this ticket.

rossengeorgiev, 2017-03-07:
https://github.com/rossengeorgiev/aprs-python/blob/9cee09b7a891b84d0c5b57991897015b1a8f2124/aprslib/parsing/position.py#L187

This has a bit about ambiguity, which won't apply to NMAE stuff.

kb1lqc, 2017-03-09:
OK so I take it this means it's sorta accounted for in RX parsing. Would I be correct in saying adding a NMEA to DD function in utils would be beneficial? I'd gladly [clean this FaradayRF code up](https://github.com/kb1lqc/Faraday-Software/blob/aprslib/Applications/APRS/aprs.py#L170) and do a PR in utils to give you the function too.

kb1lqc, 2017-03-23:
I'm closing this issue as the following was required:

* Send APRS strings with aprslib APRS-IS connection, do not just use aprslib strings and send them over a socket to APRS-IS
* Converted coordinates form NMEA to DD format prior to sending to aprslib. Double-work for Python but it's a pretty tiny amount of work in the end.

closed 2017-03-23 by kb1lqc