rossengeorgiev/aprs-python #26: Add TelemetryReport packet Generation Function

kb1lqc, 2017-03-07:
I have successfully implemented your [PositionReport()](https://github.com/kb1lqc/Faraday-Software/blob/aprslib/Applications/APRS/aprs.py#L204) function into my @FaradayRF APRS application to generate and send position reports to APRS-IS.

Example:
[image]

## TelemetryReport
By extension, it would be nice to support a telemetry packet with aprslib. Would you like me to attempt to model a telemetry generation function off of your PositionReport() to add in this functionality with a PR? I would like to incorporate that into @FaradayRF code as well.

rossengeorgiev, 2017-03-07:
Sure, give it a go. I might make some changes in the weekend as those bits were more a prototype. It would be nice to have validation for parameters (and docs).

kb1lqc, 2017-03-11:
I've committed PR #27 and #28 to add both telemetry value and telemetry unit/label support to `aprslib`. Would love to fix them up/document them enough to get the PR's merged! Will be working on telemetry parameters and scaling in the near future too.

Let me know if you like these on separate PR's or just one big one. #28 has the changes from #27 listed as well but not sure if that will end up going away as #27 is merged and pulled into the branch #28 is based on.

kb1lqc, 2017-03-13:
@rossengeorgiev I commented on my changes for #27 and updated that branch `faradayrfcustom`.

Also, I have several other branches with updates of new packets but ran into some Git hell and ended up just making a new branch `[faradayrfcustomall](https://github.com/kb1lqc/aprs-python/tree/faradayrfcustomall)` that combines all new packet types from the branches using a known good commit. My bad. Learning.

That branch has several new packet types in aprslib/packets, check it out. It does NOT have the recent updates from #27 as everytime I merge or rebase those updates into my `faradaycustomall` branch it deletes my new packet types... Suggestions? 

I'll probably open up another pull request to get these into aprslib as well. When all these functions are pulled into aprslib this rounds out my needs (initially) and lets me able to actually release master code on FaradayRF Software that uses aprslib.

kb1lqc, 2017-03-23:
Closing as @rossengeorgiev committed PR #27 with my original PR branch updated to fit his `aprslib` style. Thanks!

closed 2017-03-23 by kb1lqc