rossengeorgiev/aprs-python #29: Telemetry Parser

EA5AER, 2017-03-15:
Hi:

I've seen that the packets of telemetry are still not supported for parse. For any reason? I say that because in telemetry.py has already parse_comment_telemetry that will do more or less the job.

If a force parse_comment_telemetry for this beacon:
DEBUG:aprslib.parsing:Parsing: **ED5YAM>APTT4,EA5RCD-15*,WIDE1,WIDE2-1:T#790,551,564,999,085,716,11000000**
DEBUG:aprslib.parsing:Parsed ok.

i will obtain this:
**{'raw': u'ED5YAM>APTT4,EA5RCD-15*,WIDE1,WIDE2-1:T#790,551,564,999,085,716,11000000', 'via': '', 'from': u'ED5YAM', 'to': u'APTT4', 'path': [u'EA5RCD-15*', u'WIDE1', u'WIDE2-1'], 'format': 'beacon', 'text': u'T#790,551,564,999,085,716,11000000'}**

This is better than obtain:
Mar 15 14:42:54 MonitorAPRS sh[563]: File "/usr/local/lib/python2.7/dist-packages/aprslib/parsing/__init__.py", line 162, in _try_toparse_body
Mar 15 14:42:54 MonitorAPRS sh[563]: raise UnknownFormat("format is not supported")

rossengeorgiev, 2017-03-15:
Comment telemetry format is different. Telemetry report is not implemented, since I never needed it. I may implemented it soon™

Charlie-Gale, 2021-04-03:
It looks like something was done to address this, but when I try to use it nothing is returned:
DEBUG:aprslib.parsing:Parsing: VY1CC-3>APOT30,WIDE2-2,qAR,VY1YC-1:T#057,487,000,000,000,000,00000000
then nothing else, not even an exception. 
Compared to:
DEBUG:aprslib.parsing:Parsing: VY1YC-1>APMI06,TCPIP*,qAS,VY1RC:@031728z6045.13N/13505.86W-WX3in1Plus2.0 U=16.3V,T=0.0C
DEBUG:aprslib.parsing:Parsed as normal position report
DEBUG:aprslib.parsing:Parsed ok.

commit b0c29e45 (2017-03-19): add telemetry report parsing; fix #29
commit 62e064d2 (2017-03-19): add telemetry report parsing; fix #29
commit f04e2cf8 (2017-09-23): add telemetry report parsing; fix #29
commit 2a222e69 (2019-06-02): add telemetry report parsing; fix #29