rossengeorgiev/aprs-python #30: BASE91 Conversion Creates Invalid Value for Decimal 0 Telemetry

kb1lqc, 2017-09-05:
When using the base91 `from_decimal` function with an input of 0 the output is "!". However, APRS.fi which follows the [BASE91 comment telemetry](http://he.fi/doc/aprs-base91-comment-telemetry.txt) specification should actually create a "!!":

[image]

This is rejected by aprs.fi:

Sequence = 0, Channel 1 (ADC0) = 0
[image]

This turns into:
[image]

However, the specification clearly intends for sequence of 0 with channel 1 = 0 to be "|!!!!|"

[image]

kb1lqc, 2017-09-05:
This can be mitigated with the rjust() string method

`str.rjust(2,"!")`

This pads the left with a "!" up to two characters long which is the specification.

rossengeorgiev, 2017-09-05:
Hi @kb1lqc, the function has a second parameter to specify the width.

https://github.com/rossengeorgiev/aprs-python/blob/master/aprslib/base91.py#L54

```python
base91.from_decimal(0, width=2)

kb1lqc, 2017-09-05:
Oh snap. I can't believe I didn't see that. I was looking right at it. Must be time to get some sleep 😴 . Thanks for quickly pointing that out!

closed 2017-09-05 by kb1lqc