rossengeorgiev/aprs-python #31: Problem when Parsing Messages that include Emojies

OE8HSR, 2017-09-07:
hello.
i am new to this so please forgive me if i am doing something wrong.

i wanted to write a script that receives aprs messages and working with them.
i encountered a problem when a message comes in that contains emojies.

the error message is:

```
APRS Packet: b'SQ6VRS>APDR14,TCPIP*,qAC,T2EUSKADI:=5029.09N/01719.96E[293/001/027.280MHz/A=000902 SQ6VRS-9 \xf0\x9f\x9a\x9b.:Redzi:. MobileGSM: +48502075027'
Traceback (most recent call last):
  File "C:/Users/hmst/Desktop/aprs.py", line 27, in <module>
    client.consumer(receivepackage)
  File "C:\Users\hmst\AppData\Local\Programs\Python\Python36-32\lib\site-packages\aprslib\inet.py", line 185, in consumer
    callback(self._parse(line))
  File "C:/Users/hmst/Desktop/aprs.py", line 23, in receivepackage
    print(formatstring)
UnicodeEncodeError: 'UCS-2' codec can't encode characters in position 67-67: Non-BMP character not supported in Tk
```

is there any solution to this problem? i tried to filter these messages out but i noticed that it fails before i can filter the message...

thanks alot
Hans

rossengeorgiev, 2017-09-07:
UCS-2 probably doesn't cover the emoji range

OE8HSR, 2017-09-07:
thank you for the fast answer.
is there a way i can catch that exception so that the script doesnt shuts down?
it would be no problem when messages with that kind for "characters" would be thrown away...

could you please give me a hint where in the module the problem occurs? i have searched alot today but i havent found it. 

best wishes and thanks alot
Hans

rossengeorgiev, 2017-09-07:
Try `print(formatstring.encode('utf-16', errors='replace'))`

This way the invalid codepoints will be replaced, so that the string can be printed in your environment. Or set your environment to use UTF-8.

OE8HSR, 2017-09-07:
thank you for your fast answer.
i will try that.
honestly i never thought that someone was using emojies in APRS messages but there are more than you would guess :)

best wishes
hans

rossengeorgiev, 2017-09-23:
Closing this as it looks resolved

closed 2017-09-23 by rossengeorgiev