rossengeorgiev/aprs-python #36: Message Id (msgNO) is being concatenated

bgmarble, 2017-11-03:
I just started using YAAC to test my application. Using aprs-lib I receive a message from a YAAC client with an alphanumeric 'msgNo' or message id like this, 'C0001'. Aprslib converts it to the '1'. Here is the raw packet.

[KG7AFQ-8>APJYC1,TCPIP*,qAC,T2SJC::BAKERSMS :?{C0001]

It should create a 'msgNo' like this 'C0001'.

Would you look at this?
Thanks

rossengeorgiev, 2017-11-04:
Well yeah, that field is defined as Message Number in the protocol spec. `C0001` is not a number. The code needs to be fixed to throws a `ParseError` here.

rossengeorgiev, 2018-01-20:
I was wrong. Reading the spec `msgNo` should be alphanumeric

closed 2018-01-20 by rossengeorgiev

stroobandt, 2019-10-05:
This is still not in the latest PyPI release of version 0.6.46.

rossengeorgiev, 2019-10-08:
Hi @stroobandt, try installing the package directly from the repo

commit 94b89a6d (2018-01-20): parsing: fix incorrect parsing of msgNo

msgNo should accept alphanumeric characters

fix #36