rossengeorgiev/aprs-python #38: unexpected keyword passcode

bhhoyer, 2017-11-29:
Hi,

Thanks for providing this library.

Using the example for send a packet:

import aprslib
# a valid passcode for the callsign is required in order to send
AIS = aprslib.IS("K7UDR", passcode="16273", port=14580)
AIS.connect()
# send a single status message
AIS.sendall("K7UDR>APRS,TCPIP*:>status text")

pi@WSPi:~/python/stuart $ python send.py 
Traceback (most recent call last):
  File "send.py", line 11, in <module>
    AIS = aprslib.IS("K7UDR", passcode="16273", port=14580)
TypeError: __init__() got an unexpected keyword argument 'passcode'

bhhoyer, 2017-11-29:
figured it out deleted passcode=

closed 2017-11-29 by bhhoyer

rossengeorgiev, 2017-11-29:
Hi @bhhoyer, it's `passwd`. See http://aprs-python.readthedocs.io/en/latest/examples.html#passcodes