rossengeorgiev/aprs-python #40: What do we pass for the passwd?

wgaylord, 2018-01-31:
I have been trying to get the APRS-IS conencting with a password. But it keeps telling me my password is wrong? 

If there some specific password we have to use?  Or is this because I am using the default server. (I have an account with aprs.if but I can't find a link for if they have a server url for APRS-IS)

wgaylord, 2018-01-31:
I am a fool. xD Is the passcode  function its not an actual password...

closed 2018-01-31 by wgaylord

rossengeorgiev, 2018-01-31:
You don't need a password if you just want to receive.
Password is based on your callsign and it is required for sending.
Make sure you have read http://www.aprs-is.net/Default.aspx and the docs before attempting to send stuff.