rossengeorgiev/aprs-python #42: Parsing has extra u in front of value

wsmitchell3, 2018-02-05:
When parsing, many of the values in the dictionary have a u before the quoted value.
Input:
`W0DRK-2>APRS,KD0BXJ-2,N0PBA-1,WIDE2*:!4409.98N/09537.30W# W0DRK Digi Tracy MN<0x20>`

Yields output:
`{'comment': u'W0DRK Digi Tracy MN<0x20>', 'via': '', 'from': u'W0DRK-2', 'posambiguity': 0, 'to': u'APRS', 'messagecapable': False, 'symbol': u'#', 'format': 'uncompressed', 'longitude': -95.62166666666667, 'raw': u'W0DRK-2>APRS,KD0BXJ-2,N0PBA-1,WIDE2*:!4409.98N/09537.30W# W0DRK Digi Tracy MN<0x20>', 'symbol_table': u'/', 'latitude': 44.166333333333334, 'path': [u'KD0BXJ-2', u'N0PBA-1', u'WIDE2*']}`

rossengeorgiev, 2018-02-05:
Yes, the strings are returned as `unicode` in python2. What is the issue?

wsmitchell3, 2018-02-06:
Ah.  Okay, I hadn't seen that notation before.  I have it exporting correctly into the next part of my program now.  Thanks.

closed 2018-02-06 by wsmitchell3