rossengeorgiev/aprs-python #50: Packets not being returned by parser

ivanmartinezmorales, 2019-03-19:
Hi, when attempting to stream packets with a specific callsign, I am unable to stream said packet. The script hangs up at 
```python 
def callback(packet):
    return packet 
```
Where it does not exit the loop. I previously had it working, where I could process, parse, and return positioning data from aprslib, but recently it has stopped working. I have tried using the non-blocking feature for the .consumer() method, but also had no success with it.

rossengeorgiev, 2019-03-23:
Hi, you cannot just return in the callback as that will go nowhere. See the code that calls the callback https://github.com/rossengeorgiev/aprs-python/blob/94b89a6da47a322129484efcaf1e82f6a9932891/aprslib/inet.py#L182-L185

You will need do something like adding it to a queue, printing, writing to a file, or anything else.

ivanmartinezmorales, 2021-10-12:
Hi, sorry to leave this issue open for so long. Thanks for answering my question!

closed 2021-10-12 by ivanmartinezmorales