rossengeorgiev/aprs-python #51: Question about coordinates conversion

sp2ong, 2019-05-07:
How to use aprslib to convert compressed coordinates to decimal format?

lat=aprslib.xxxxx("4903.50N")
long=aprslib.xxxx("07201.75W")

rossengeorgiev, 2019-05-07:
Hi @sp2ong, there isn't a function to do it. Here is the relevant code:

https://github.com/rossengeorgiev/aprs-python/blob/94b89a6da47a322129484efcaf1e82f6a9932891/aprslib/parsing/position.py#L99-L100

sp2ong, 2019-05-07:
Thank you for replay. I have try use

import aprslib
form aprslib import base91

lat = 90 - (base91.to_decimal("5303.52N") / 380926.0) 
print lat

but result is strange:

-2740265826.18

rossengeorgiev, 2019-05-07:
That is not compressed. That is just degrees, minutes and direction. There is no convenient function available to convert them to decimal format. The relevant code is here:

https://github.com/rossengeorgiev/aprs-python/blob/94b89a6da47a322129484efcaf1e82f6a9932891/aprslib/parsing/position.py#L132-L192

sp2ong, 2019-05-07:
Ok, thank you for repaly

closed 2019-05-07 by sp2ong