rossengeorgiev/aprs-python #52: [question] how to extract parsed value ftom the dictionary

iw5ejm, 2019-08-07:
`def callback(packet):
    print(packet)
    T = packet.get('temperature')
    print(T)`

Running this callback function on a weather packet filtered stream, with AIS.consumer(callback, raw=False)

Why I'm getting only KeyError Exceptions  or None value for the temperature (and for all the other key in the dict)? It seems that there is no key in packet.. What am I doing wrong?

thanks,
Marco

iw5ejm, 2019-08-07:
problem was the nested dictionary for weather data

you have to use something like this to get wx data:

>     T = packet['weather']['temperature']
>     print(T)
>     H = packet['weather']['humidity']
>     P = packet['weather']['pressure']

closed 2019-08-07 by iw5ejm

rossengeorgiev, 2019-08-08:
Hi @iw5ejm, the packet will be a `dict` and will have a different structure depending on the message. You will have to check for the `weather` key, or that format is `wx`. You can see the various package structure in the docs: https://aprs-python.readthedocs.io/en/latest/parse_formats.html#weather