rossengeorgiev/aprs-python #72: Missing wind speed / direction weather parse failure

powturns, 2022-03-15:
Weather information is not parsed from the following packets:

```
CW1129>APRS,TCPXX*,qAX,CWOP-4:@132350z4235.56N/07123.21W_.../000g000t030r000p000P000b10149h33.weewx-4.5.1-Vantage

CW1604>APRS,TCPXX*,qAX,CWOP-4:@132345z4444.70N/06531.17W_.../...g...t031r000p010P002h58b10156L....DsIP

'CW1367>APRS,TCPXX*,qAX,CWOP-5:@152159z4026.39N/07406.71W_.../...g...t065r000p000P000b10163h33eMB51'
```

I suspect it's due to the missing wind speed / direction, however these values are optional according to APRS101 pg 27.

aprslib ~= 0.7.1

rossengeorgiev, 2022-07-10:
No, `.../000` is actually data extension for course and speed. They are empty, because the station is static. APRS101 pg 33.

closed 2022-07-10 by rossengeorgiev

rossengeorgiev, 2022-07-10:
Fixed in `0.7.2`.

```python
In [1]: import aprslib

In [2]: aprslib.__version__
Out[2]: '0.7.2'

In [3]: aprslib.parse("CW1129>APRS,TCPXX*,qAX,CWOP-4:@132350z4235.56N/07123.21W_.../000g000t030r000p000P000b10149h33.weewx-4.5.1-Vantage")
Out[3]:
{'raw': 'CW1129>APRS,TCPXX*,qAX,CWOP-4:@132350z4235.56N/07123.21W_.../000g000t030r000p000P000b10149h33.weewx-4.5.1-Vantage',
 'from': 'CW1129',
 'to': 'APRS',
 'path': ['TCPXX*', 'qAX', 'CWOP-4'],
 'via': 'CWOP-4',
 'messagecapable': True,
 'raw_timestamp': '132350z',
 'timestamp': 1657756200,
 'format': 'uncompressed',
 'posambiguity': 0,
 'symbol': '_',
 'symbol_table': '/',
 'latitude': 42.592666666666666,
 'longitude': -71.38683333333333,
 'comment': '.weewx-4.5.1-Vantage',
 'weather': {'wind_gust': 0.0,
  'temperature': -1.1111111111111112,
  'rain_1h': 0.0,
  'rain_24h': 0.0,
  'rain_since_midnight': 0.0,
  'pressure': 1014.9,
  'humidity': 33}}

In [4]: aprslib.parse("CW1604>APRS,TCPXX*,qAX,CWOP-4:@132345z4444.70N/06531.17W_.../...g...t031r000p010P002h58b10156L....DsIP")
Out[4]:
{'raw': 'CW1604>APRS,TCPXX*,qAX,CWOP-4:@132345z4444.70N/06531.17W_.../...g...t031r000p010P002h58b10156L....DsIP',
 'from': 'CW1604',
 'to': 'APRS',
 'path': ['TCPXX*', 'qAX', 'CWOP-4'],
 'via': 'CWOP-4',
 'messagecapable': True,
 'raw_timestamp': '132345z',
 'timestamp': 1657755900,
 'format': 'uncompressed',
 'posambiguity': 0,
 'symbol': '_',
 'symbol_table': '/',
 'latitude': 44.745,
 'longitude': -65.5195,
 'comment': '.DSIP',
 'weather': {'temperature': -0.5555555555555556,
  'rain_1h': 0.0,
  'rain_24h': 2.54,
  'rain_since_midnight': 0.508,
  'humidity': 58,
  'pressure': 1015.6}}

In [5]: aprslib.parse('CW1367>APRS,TCPXX*,qAX,CWOP-5:@152159z4026.39N/07406.71W_.../...g...t065r000p000P000b10163h33eMB51')
Out[5]:
{'raw': 'CW1367>APRS,TCPXX*,qAX,CWOP-5:@152159z4026.39N/07406.71W_.../...g...t065r000p000P000b10163h33eMB51',
 'from': 'CW1367',
 'to': 'APRS',
 'path': ['TCPXX*', 'qAX', 'CWOP-5'],
 'via': 'CWOP-5',
 'messagecapable': True,
 'raw_timestamp': '152159z',
 'timestamp': 1657922340,
 'format': 'uncompressed',
 'posambiguity': 0,
 'symbol': '_',
 'symbol_table': '/',
 'latitude': 40.43983333333333,
 'longitude': -74.11183333333334,
 'comment': 'eMB51',
 'weather': {'temperature': 18.333333333333332,
  'rain_1h': 0.0,
  'rain_24h': 0.0,
  'rain_since_midnight': 0.0,
  'pressure': 1016.3,
  'humidity': 33}}
```

commit 2a27dc44 (2022-07-10): parse_position: fix data ext and weather parsing

Fix #72