rossengeorgiev/aprs-python #73: PHG and comment parsing error

itec78, 2022-07-10:
Packets with "phg" info, are parsed wrong.
example: "IR4BA>APMI06,IR5X*,WIDE2-1,qAR,IK5UIK-3:!4413.07N/01105.57E#PHG2830Balzo S.Caterina Group"
is parsed as:  'phg': '2830B', 'comment': 'alzo S.Caterina Group'
it should be:  'phg': '2830', 'comment': 'Balzo S.Caterina Group'
thanks

rossengeorgiev, 2022-07-10:
Are you running the latest version `0.7.1`? `python -c 'import aprslib; print(aprslib.__version__)'` 

```python
In [3]: parse("IR4BA>APMI06,IR5X*,WIDE2-1,qAR,IK5UIK-3:!4413.07N/01105.57E#PHG2830Balzo S.Caterina Group")
Out[3]:
{'raw': 'IR4BA>APMI06,IR5X*,WIDE2-1,qAR,IK5UIK-3:!4413.07N/01105.57E#PHG2830Balzo S.Caterina Group',
 'from': 'IR4BA',
 'to': 'APMI06',
 'path': ['IR5X*', 'WIDE2-1', 'qAR', 'IK5UIK-3'],
 'via': 'IK5UIK-3',
 'messagecapable': False,
 'format': 'uncompressed',
 'posambiguity': 0,
 'symbol': '#',
 'symbol_table': '/',
 'latitude': 44.21783333333333,
 'longitude': 11.092833333333333,
 'phg': '2830',
 'phg_power': 4,
 'phg_height': 780.288,
 'phg_gain': 1.9952623149688795,
 'phg_dir': 'omni',
 'phg_range': 91.5251154073474,
 'comment': 'Balzo S.Caterina Group'}
```

itec78, 2022-07-10:
No, I was running an old version. I forgot to check it before opening the issue. Thanks!

closed 2022-07-10 by itec78