rossengeorgiev/aprs-python #74: Comment parsing

itec78, 2022-07-10:
Hi.
It seems that sometimes parser adds extra char to comment

IZ4FTD-9>TS4R21,WIDE1-1,WIDE4-4,qAR,IZ5RAC-3:'&P& p~>/]"4&}JEEPCOMP>qrv:RU15-Maurizio
is parsed as 'comment': ']JEEPCOMP>qrv:RU15-Maurizio'

IZ4TNW-9>TT1W86,WIDE1-1,WIDE2-2,qAR,IR4BA:`']LnRL>/]"8W}=
is parsed as 'comment': ']='

Could you please check?

rossengeorgiev, 2022-07-10:
Both look correctly parsed to me. Here is how aprs.fi decodes them:

https://aprs.fi/?c=raw&call=IZ4FTD-9&limit=100&view=decoded
```
2022-07-08 09:27:59 BST: IZ4FTD-9>TS3X05,WIDE1-1,WIDE4-4,qAR,IR5X:'&>| fu>/]"4&}JEEPCOMP>qrv:RU15-Maurizio
   type: location
   format: mice
   srccallsign: IZ4FTD-9
   dstcallsign: TS3X05
   latitude: 43.634166666667 °
   longitude: 10.582666666667 °
   course: 89 °
   speed: 87.044 km/h
   altitude: 15 m
   symboltable: /
   symbolcode: >
   mbits: 110
   posresolution: 18.52 m
   posambiguity: 0
   comment: ]JEEPCOMP>qrv:RU15-Maurizio
```

`]"4&}J...` and `]"8W}=` are practically the same, and end up in the comment as `]J...` and `]=` respectively.

itec78, 2022-07-10:
Comment in this page https://aprs.fi/info/IZ4FTD-9 was without "]", so I thought it was a parsing error. Thanks a lot for your answer!

closed 2022-07-10 by itec78